tomate44/CurvesWB #5: Screenshots + other things

luzpaz, 2017-05-24:
Thanks for your work on this addon. I'm volunteering to improve the addon manager. Would you mind:

- [ ] adding some screenshots of your addon  
- [ ] tagging this repo with the 'freecad' tag  
- [ ] stating what freecad versions this addon is compatible with  
- [ ] if this is compatible with latest 0.17 version please inform users they can now download addon through the new addon manager

Thanks!

tomate44, 2017-05-24:
Hi,
Indeed my addon is totally undocumented.
It is time for a little bit of work in this area.
I'll do it when I get some time.
Thanks for your work on the bugtracker ( and now the addon manager !).

luzpaz, 2017-05-25:
👍 
I'm happy to help as well. Just point me to some more info.

luzpaz, 2018-09-02:
@tomate44 can you add https://forum.freecadweb.org/viewtopic.php?f=8&t=22675 to the README as the forum thread for people to post on to ask/discuss all things CurvesWB ?

closed 2018-09-03 by luzpaz