tomate44/CurvesWB #34: Error message logged at each FreeCAD startup

0penBrain, 2020-11-15:
Each time I start FreeCAD, Curves WB throws following error in log :
`During initialization the error ""'CurvesWorkbench' already exists."" occurred in freecad.Curves
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 154, in InitApplications
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/user/.FreeCAD/Mod/Curves/freecad/Curves/init_gui.py", line 143, in <module>
    Gui.addWorkbench(CurvesWorkbench())
KeyError: "'CurvesWorkbench' already exists."
--------------------------------------------------------------------------------`

tomate44, 2020-11-20:
CurvesWB switched to new "namespace workbench" file structure some months ago.
Maybe you have files from the old CurvesWB file structure conflicting.
Could you delete the CurvesWB folder and install again, please ?

0penBrain, 2020-11-20:
@tomate44 : you got it right. No more issue after de-installing + re-installing.
The weird thing is that I installed/upgraded only through the addon manager. Maybe there is something to improve there.
Anyway, thanks and sorry for the noise.

tomate44, 2020-11-20:
I think I didn't take care of suppressing the old files in the switch commit ...

closed 2020-11-20 by tomate44