tomate44/CurvesWB #42: Translate Curves Workbench

luzpaz, 2021-02-19:
https://wiki.freecadweb.org/Translating_an_external_workbench  
https://github.com/FreeCAD/FreeCAD-addons/issues/187

luzpaz, 2023-11-28:
@tomate44 would you be ok if someone spearheaded this ?

luzpaz, 2023-11-28:
CC @kaktusus

tomate44, 2023-11-29:
I am not in favor of adding translation to the workbench right now.
The workbench is not stable. I don't want to have to deal with translation burden during development.
I don't think CurvesWB will ever become stable.
I think a better approach will be to develop a new workbench that picks up the most useful and stable tools from Curves.
This one would be a good candidate for translation.

kaktusus, 2023-11-29:
CurvesWB is a really great workbench.
And it is worth getting translations one day.
But if you think it's too early for that, it's a good idea to wait.

luzpaz, 2024-09-10:
@tomate44 there are many users of Curves WB that means it's popular. Where do we go from here? Wait for new workbench or wait for Surface WB to be developed ?