tomate44/CurvesWB #55: Reverse flag in Blend curve doesn't honor parameter set to 0 mm

janbraiins, 2021-11-28:
I have come across a small issue. My scenario:

- I want to connect 2 curves with the blend curve feature
- I want to use 0 mm for both parameters, so that the blended curve is agnostic to changes in lengths of both curves
- This requires having one of the ends set in 'reverse' direction
The problem:
- setting the parameter 0 mm doesn't switch the blend point to the other end of the curve = 0mm always means only one end of the curve, which I believe is not expected.
I am attaching a small example where I have set the parameter 1 to 0.001 mm which actually works. Try setting it 0, it would jump to the other end of the curve.

Still want to thank you for this excellent workbench.

Jan

Below are screenshots of the scenario with the 0.001 mm workaround
[image]
[image]
Switching to 0 mm:
[image]
[image]

tomate44, 2021-12-01:
It should be fix with [commit](https://github.com/tomate44/CurvesWB/commit/6a1cf9c4f179d7aee7ebbc3638e0ac448764696b). 
Thanks for the report.
I close.

closed 2021-12-01 by tomate44

janbraiins, 2021-12-02:
thank you for a swift fix ;-)