tomate44/CurvesWB #56: "Shape already exists" error while using the "profile support plane" tool.

markus61, 2021-12-11:
22:44:00  Running the Python command 'profileSupportCmd' failed:
Traceback (most recent call last):
  File "/home/markus/.local/share/FreeCAD/Mod/Curves/freecad/Curves/ProfileSketch.py", line 143, in Activated
    self.makeProfileFeature(shapes, params)
  File "/home/markus/.local/share/FreeCAD/Mod/Curves/freecad/Curves/ProfileSketch.py", line 105, in makeProfileFeature
    profileSupportFP(prof)
  File "/home/markus/.local/share/FreeCAD/Mod/Curves/freecad/Curves/ProfileSketch.py", line 33, in __init__
    obj.addProperty("Part::PropertyPartShape","Shape",       "Profile",   "Shape of the object")

Property runen#Profile001.Shape already exists

tomate44, 2021-12-13:
Hi,
Thanks for the report.
Should be fixed with [latest commit](https://github.com/tomate44/CurvesWB/commit/c917c51159942fa1af7069b52cc964736af7dde2).

closed 2021-12-13 by tomate44

markus61, 2021-12-13:
thanx for the fix!
worx like a charm now :-)

/markus