tomate44/CurvesWB #57: Workbench does not load (python 3.10?)

dpellegr, 2022-01-02:
Hi, the workbench does not want to load. Here is the stacktrace. It might be related to python v3.10.

```
19:10:32  <built-in function SoFieldContainer_getField> returned a result with an exception set
19:10:32  Traceback (most recent call last):
  File "/home/dario/.FreeCAD/Mod/Curves/freecad/Curves/init_gui.py", line 20, in Initialize
    graphics.Marker([App.Vector()])
  File "/home/dario/.FreeCAD/Mod/Curves/freecad/Curves/graphics.py", line 152, in __init__
    super(Marker, self).__init__(points, dynamic)
  File "/home/dario/.FreeCAD/Mod/Curves/freecad/Curves/graphics.py", line 63, in __init__
    self.set_color()
  File "/home/dario/.FreeCAD/Mod/Curves/freecad/Curves/graphics.py", line 88, in set_color
    self.color.diffuseColor = COLORS[self.std_col]
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3998, in __setattr__
    field = self.getField(name)
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3873, in getField
    return _coin.SoFieldContainer_getField(self, name)
```

dpellegr, 2022-01-03:
It probably comes from this: https://github.com/coin3d/pivy/issues/88
Python 3.10 is creating a lot of issues in freecad, but the root is deep and there is probably nothing to adjust at the workbench level.

tomate44, 2022-02-17:
I think it is now fixed in pivy / coin3d

dpellegr, 2022-02-17:
> I think it is now fixed in pivy / coin3d

You are right. Things have been working for a while now and I forgot about this issue.