tomate44/CurvesWB #63: [FR] increase (Select adjacent faces)

hamza00789, 2022-02-03:
Point Selection   

usage 
  
All connected faces will be selected from the point selection (Tangent or non Tangent) 
 
Example 

[image] 

[image]
      
Tangent Selection   

usage 

In the tangent selection, all connected faces that are tangent with each other will be selected. 

Example         

[image] 

[image]

Invert selection   
 
usage 

The invert selection will select all faces except the one selected. 

Example 

[image] 

[image]

luzpaz, 2022-12-21:
Can you x-post to the [dedicated forum thread](https://forum.freecadweb.org/viewtopic.php?f=8&t=22675) and x-post the URL back here? TIA

luzpaz, 2023-02-26:
@hamza00789 please stop closing and reopening this ticket.

closed 2024-10-12 by tomate44