tomate44/CurvesWB #67: Sketch on surface can't handle Bezier

fancsali, 2022-03-29:
When mapping a sketch on a curved surface whenever the sketch contains a Bezier curve, the projection seems to fail, and the sketch "shrinks": suddenly things that span the whole blue frame won't span the actual surface.

**Steps to reproduce**:
1. Create a curved shape, like a Cylinder with an angle of 180 &deg;
2. Select the curved surface, and map a (brand new) sketch onto it with the Curves WB's "Sketch on Surface" button
3. Draw a Bezier curve with the starting point coinciding with one corner of the blue frame and the endpoint coinciding with the one diagonally opposite (with a few points in between).
4. Observe, that the mapped curve does not reach the corners. (See middle cylinder in illustration)

**Expected behaviour**:
The curve's endpoints reach the corners of the curved outer surface. Similar to the case of drawing the diagonal of the blue frame using a simple line, as on the left cylinder in the picture.

**Further details**:
In the case of the right cylinder in the picture there's one diagonal drawn using a line and the other one using a Bezier curve (with 2 intermediate points between the endpoints); and that breaks down too. Hence my hypothesis is that including any Bezier curve in the sketch will break the mapping for the whole sketch.

[image]

tomate44, 2022-03-30:
Hi,
This happens because the circles around the poles in the sketch curves are also used for the bounding box computation.
To avoid this, in sketch edit mode, you can select the Bezier curve and do Sketch -> Sketcher tools -> Show / hide internal geometry.

closed 2022-03-30 by tomate44

fancsali, 2022-03-30:
I'd still argue, this is rather counter-intuitive. Can't we skip those for the bounding box logic?

In the meantime I'll try to get this into the docs at least.