tomate44/CurvesWB #71: Join Curves fail but is fine on older versions

mangojellysolutions, 2022-05-14:
Hi
I have tried the following on both freeCAD 0.20 and 0.19 and
I have tried this on a older version of the Curves workbench and it works fine.  When I updated the workbench today and tried running a join curve between two lines (tried sketch edges, freehand curves, etc) I get the error message:

18:52:18  Traceback (most recent call last):
  File "/home/oranges/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/JoinCurves.py", line 162, in execute
    obj.Shape = obj.ExtensionProxy.approximate(obj, outEdges)
<class 'TypeError'>: type must be 'Shape', not list

This happens on both version of freeCAD.  I am running on Ubuntu

Thanks

tomate44, 2022-05-15:
Hello,
Indeed, I introduced a bug with my latest commit.
It should be fixed now.
BTW : Thanks a lot for all your great YT videos about the Curves Workbench.

tomate44, 2022-10-11:
Yes, that would be great if I could contact you on FreeCAD forum, because I have no Google account, for ethical reasons.
BTW, I have a new surfacing tool that I hope to release soon.