tomate44/CurvesWB #72: Blend Surface no working in Thunderbird development branch

buzeceji89, 2022-07-02:
The report:
```
09:55:18  <App> blendSurfaceFP_new.py(121)|GeoFeatureGroupExtension.cpp(575): Furnace#Blend_Surface.Sources(links to Furnace#Wrap001) is out of scope:Furnace#Body vs. <global>
09:55:18  Traceback (most recent call last):
  File "/home/user/.FreeCAD/Mod/Curves/freecad/Curves/blendSurfaceFP_new.py", line 67, in execute
    bs = blend_curve.BlendSurface(edges[0], faces[0], edges[-1], faces[-1])
<class 'IndexError'>: list index out of range
09:55:18  7.45e-07 <App> blendSurfaceFP_new.py(121)|Document.cpp(4266): Failed to recompute Furnace#Blend_Surface: list index out of range
09:55:18  Recompute failed! Please check report view.
```

luzpaz, 2022-12-21:
Can you x-post to the [dedicated forum thread](https://forum.freecadweb.org/viewtopic.php?f=8&t=22675) and x-post the URL back here? TIA

luzpaz, 2023-11-29:
@buzeceji89 can you still reproduce?

closed 2024-09-29 by tomate44