tomate44/CurvesWB #73: Translation of GUi

kaktusus, 2022-07-04:
How likely is this workbench to join the FreeCAD-addons on the Crowdin platform?

I am interested in the author's opinion in this regard.

kaktusus, 2023-05-24:
A developer's manual was created:

https://github.com/FreeCAD/DevelopersHandbook/blob/master/technical/translation.md

hasecilu, 2024-10-14:
Hi @tomate44, I was looking to implement the translation support on this WB but I encountered some nuisances that should be fixed before:
- [ ] Not all commands use `Curves_` as prefix, making this change will enable users to use `What's this?` command on commands and get the correct Wiki article opened. Actually I would want that commands take name as in the Wiki documentation.
- [ ] Some properties' tooltip is "Tooltip", something more descriptive is better.
- [ ] The project don't have any code formatting tools config file that tells me the desired format, since the changes would create longer lines, these would be to be formatted, Main FreeCAD project uses black with 100 char line width, I like it: https://github.com/FreeCAD/FreeCAD/blob/main/.pre-commit-config.yaml

I can make all these changes before making the translation if you want, let me know what you think.