tomate44/CurvesWB #74: bug of Sketch_on_Surface - straight lines missing, only splines are ok

rejnok93, 2022-09-16:
For this certain project (FCStd file attached) - trying to put Text on Cilinder, but result is wrong, all straight lines are missing:
[image]
Original text is  `D+T ♥ 15.7.2017`
[darek_vaza_textonly2_sosbad.FCStd.zip](https://github.com/tomate44/CurvesWB/files/9586681/darek_vaza_textonly2_sosbad.FCStd.zip)
[Arial.ttf.zip](https://github.com/tomate44/CurvesWB/files/9586730/Arial.ttf.zip)
copy of console log:
[image]
FreeCAD v0.20-1 (macOS 10.13.6), Curves Workbench installed from Addons Manager today (actual version)

tomate44, 2022-09-22:
It works OK here.
[image]

luzpaz, 2022-12-21:
@tomate44 [About](https://wiki.freecad.org/About) info ?

closed 2024-09-29 by tomate44