tomate44/CurvesWB #77: Blend Surface and Bland Solid error

rafaljot, 2023-02-01:
WB Blend Surface and Bland Solid  seems to be a great toolset. However, it looks like it doesn't work properly in certain situations. Several pairs of surfaces were connected correctly while one was not

[image]
[WB sample.FCStd.zip](https://github.com/tomate44/CurvesWB/files/10562766/WB.sample.FCStd.zip)

luzpaz, 2023-02-05:
please drag/drop zip file in to this issue for it to attach, thanks

rafaljot, 2023-02-06:
Sure. 
[WB sample 2.FCStd.zip](https://github.com/tomate44/CurvesWB/files/10663437/WB.sample.2.FCStd.zip)

rafaljot, 2023-02-06:
It looks as if he led the tangents to the next corners.
[image]

rafaljot, 2023-07-21:
any idea?

luzpaz, 2023-07-22:
cc @tomate44

tomate44, 2023-11-28:
Should be fixed in new version 0.6.15

closed 2023-11-28 by tomate44

rafaljot, 2023-11-28:
Yes, now it works.

[image]