tomate44/CurvesWB #78: Sketch_On_Surface can Fill Extrusion or Fill Faces, but not both

risacher, 2023-02-03:
Using Sketch_On_Surface to map paths onto a cylinder with offsets and thickness.  I can Fill Extrusion, or I can Fill Faces, but if I try to do both at once, it fails with this error:

```
13:41:48  Traceback (most recent call last):
  File "/Users/magnus/Library/Application Support/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 312, in execute
    ruled.check(True)
<class 'ValueError'>: BOP check found the following errors:
Error in Vertex: BOPAlgo SelfIntersect
Error in Vertex: BOPAlgo SelfIntersect
Error in Vertex: BOPAlgo SelfIntersect
Error in Vertex: BOPAlgo SelfIntersect
Error in Edge: BOPAlgo TooSmallEdge
Error in Edge: BOPAlgo TooSmallEdge

13:41:48  Recompute failed! Please check report view.

```
Tried with FreeCAD 0.21 (nightly), 0.20 (stable) and RealThunder's Link Branch.  The 2D geometry I'm trying to map is SVG paths imported from Inkscape.

The error message seems to indicate it's caused by self-intersecting paths, but close inspection of the path that throw the error don't seem to self-intersect.  

[freecad-ring.zip](https://github.com/tomate44/CurvesWB/files/10582626/freecad-ring.zip)

tomate44, 2023-02-04:
I pushed a fix to ignore invalid shapes.
Please give a try.

risacher, 2023-02-04:
Thank you for your prompt action!  This fixed my issue. I get a lot of diagnostic messages that make it seem like it shouldn't be working, but the result is exactly as I expect.  (Pasted below, in case it helps you in some way).  Please feel free to close this issue if you are satisfied with your fix. 

As a tangential question, should it be possible to use a JoinCurve as an "Extra Object" for Sketch_On_Surface?  I couldn't get this to work, although it seems like it should be useful.

```
16:49:55  Sketch bounds = (0.27303100000000313, 51.480991253513636, -3.925995, 0.0740050000000001)
16:49:55  Target face bounds = (0.0, 6.283185307179586, 0.0, 4.0)
16:49:55  mapping shape #  1
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    0:Invalid initial face
16:49:55  Total : 1 shapes
16:49:55  mapping shape #  2
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    0:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    1:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    2:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    3:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    4:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    5:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    6:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    7:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    8:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    9:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   10:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   11:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   12:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   13:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   14:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   15:Invalid initial face
16:49:55  Total : 17 shapes
16:49:55  mapping shape #  1
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    0:Invalid initial face
16:49:55  Total : 1 shapes
16:49:55  mapping shape #  2
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    0:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    1:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    2:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    3:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    4:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    5:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    6:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    7:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    8:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55    9:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   10:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   11:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   12:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   13:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   14:Invalid initial face
16:49:55  Self-intersecting wire
Unorientable shape

16:49:55   15:Invalid initial face
16:49:55  Total : 17 shapes
```

closed 2023-02-20 by risacher