tomate44/CurvesWB #88: Sketch_On_Surface fails with `BRepCheck_Analyzer::Init() - NULL shape`

yawkat, 2023-09-12:
When trying to extrude a sketch containing certain text characters using Sketch_On_Surface, the extrusion fails with the following log:

```
09:06:22  Sketch bounds = (-42.00254100000001, 200.946716, -31.21605500000001, 43.55298600000001)
09:06:22  Target face bounds = (0.0, 6.283185307179586, 0.0, 1.0)
09:06:22  mapping shape #  1
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    0:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    1:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    2:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    3:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    4:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    5:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    6:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    7:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    8:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    9:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   10:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   11:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   12:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   13:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   14:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   15:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   16:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   17:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   18:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   19:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   20:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   21:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   22:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   23:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   24:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   25:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   26:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22   27:Invalid initial face
09:06:22  Total : 28 shapes
09:06:22  mapping shape #  1
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    0:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    1:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    2:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    3:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    4:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    5:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    6:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    7:Invalid initial face
09:06:22  Self-intersecting wire
Unorientable shape

09:06:22    8:Invalid initial face
09:06:22  Traceback (most recent call last):
  File "/home/yawkat/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 297, in execute
    shapes_2 = self.map_shapelist(imput_shapes, quad, f2.Face1, obj.FillFaces)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yawkat/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 199, in map_shapelist
    shapelist.extend(self.map_shape(shape, quad, face, fillfaces))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yawkat/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 232, in map_shape
    return self.build_faces(wirelist, face)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yawkat/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 180, in build_faces
    f.validate()
<class 'Part.OCCError'>: BRepCheck_Analyzer::Init() - NULL shape
09:06:22  Recompute failed!
```

This issue seems to be specific to some characters such as 'k' and 'n'.

I've attached a reproducer project.

[null-shape-sample.FCStd.gz](https://github.com/tomate44/CurvesWB/files/12583439/null-shape-sample.FCStd.gz)

edwilliams16, 2023-10-20:
null-shape-sample.FCStd.gz unzips to a folder, not a FCStd file, so can't open it.

yawkat, 2023-10-20:
what do you mean? it's a gz file, it can't unzip to a folder. But I will attach it as a zip as well.

[null-shape-sample.zip](https://github.com/tomate44/CurvesWB/files/13056703/null-shape-sample.zip)

edwilliams16, 2023-10-21:
zip file works  gz didn't.

 I notice that it works with a reduced number of letters - despite the error messages. Add one more letter to the 'abcdefg' Shapestring in my file (doesn't matter which!) and it fails.  An error counter somewhere -or the wire sort finally fails?

[null-shape-sample_EW.FCStd.zip](https://github.com/tomate44/CurvesWB/files/13059672/null-shape-sample_EW.FCStd.zip)
[image]

closed 2024-04-20 by tomate44