tomate44/CurvesWB #112: Sketch on surface thickness don't work with splines

Anthony-Gaudino, 2024-01-01:
I created a Sketch on surface from a cylinder face, when I create a spline in the mapped sketch, when I add a value to the `thickness` parameter, it does not create a lateral surfaces between the top and bottom surfaces.
[image]

It works with other sketch elements:
[image]

avgeeklucky, 2024-02-05:
Not sure it's a splines issue. For me I'm getting errors on concave and convex surfaces.

I created some examples to help try and find the culprit: [curve-bug.zip](https://github.com/tomate44/CurvesWB/files/14166086/curve-bug.zip)

In the example file I get this error:
```txt
13:41:42  Traceback (most recent call last):
  File "C:\Users\Lucky\AppData\Roaming\FreeCAD\Mod\Curves\.\freecad\Curves\Sketch_On_Surface.py", line 318, in execute
    f2 = face.makeOffsetShape(obj.Offset+obj.Thickness, 1e-7)
<class 'Part.OCCError'>: BRepOffsetAPI_MakeOffsetShape not done
13:41:42  Recompute failed!
```

In another example, [thandle6.zip](https://github.com/tomate44/CurvesWB/files/14166162/thandle6.zip), with a more complex sketch (svg import) I get another error. Removing all the splines here does seem to work though.

```txt
13:46:31  <Exception> Access violation
13:46:31  Traceback (most recent call
 last):
  File "C:\Users\Lucky\AppData\Roaming\FreeCAD\Mod\Curves\.\freecad\Curves\Sketch_On_Surface.py", line 333, in execute
    ruled = Part.makeRuledSurface(shapes_1[i].Edges[j], shapes_2[i].Edges[j])
<class 'RuntimeError'>: FreeCAD exception thrown (Access violation)
13:46:31  Recompute failed!
```

Version Info:
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.23
```

andreadinale, 2024-05-21:
I can confirm, not a splines issue. If i have "Fill faces" set true, i obtain this error when i try to set thickness. With "Fill faces" set to false no error.

FreeCAD 0.21.2 and 0.21.1
Curves last master release (21/May/2024)

12:34:44  <Exception> Access violation
12:34:44  Traceback (most recent call last):
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\Curves\.\freecad\Curves\Sketch_On_Surface.py", line 358, in execute
    ruled = Part.makeRuledSurface(shapes_1[i].Edges[j], shapes_2[i].Edges[j])
<class 'RuntimeError'>: FreeCAD exception thrown (Access violation)
12:34:44  Recompute failed!

[image]

tomate44, 2024-10-10:
I cannot reproduce on the supplied files.
Please reopen and provide example files if bug is still valid.

closed 2024-10-10 by tomate44