tomate44/CurvesWB #119: Module __FreeCAD_Base__ has no attribute precision

mikeq3, 2024-04-15:
When I select the Curves Workbench for the first time after starting FreeCAD, I see the error: 
"Module '__FreeCAD_Base__' has no attribute precision"
Also, I am not seeing the toolbar for this workbench most likely due to the error.
I'm running on Fedora linux and using FreeCAD version 0.20.2
This is after installing the workbench from the add on manager in FreeCAD.

Here is the error log display:
11:46:53  module '__FreeCADBase__' has no attribute 'Precision'
11:46:53  Traceback (most recent call last):
  File "/home/mikeq/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/init_gui.py", line 38, in Initialize
    from . import IsoCurve
  File "/usr/lib64/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/mikeq/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/IsoCurve.py", line 22, in <module>
    from freecad.Curves import isocurves
  File "/usr/lib64/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/mikeq/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/isocurves.py", line 19, in <module>
    TOL3D = FreeCAD.Base.Precision.confusion()

tomate44, 2024-04-17:
It should be fixed in version 0.6.33
Can you please confirm ( I am not able to test in FreeCAD 0.20.2 ) ?

mikeq3, 2024-04-22:
Sorry I just saw this. I just updated to 0.6.35 and it looks good so far in FreeCad 0.20.2

closed 2024-04-23 by tomate44