tomate44/CurvesWB #124: Curves addon error

4567Nickam, 2024-07-02:
Many tools missing and not working at all
[Uploading curves WB errors.txt…]()

Syres916, 2024-07-03:
With FreeCAD closed down, using Windows Explorer navigate to the following folder:

_C:\Program Files\FreeCAD 0.21\FreeCAD 0.21\bin\Lib\site-packages_

Delete any subfolders that start with:
`numpy`
`scipy`

Repeat by navigating to the following two folders and delete any `numpy` or `scipy` subfolders:

_C:\Users\yourusername\AppData\Roaming\FreeCAD\AdditionalPythonPackages_
_C:\Users\yourusername\AppData\Roaming\FreeCAD\AdditionalPythonPackages\py38_

Download the attached macro into your macros folder, the default location is _C:\Users\yourusername\AppData\Roaming\FreeCAD\Macro_
Rename the file by removing the .zip off the end so it reads _RunPipOnWin.py_

Start FreeCAD, don't be alarmed if you get even more errors, that wouldn't be unexpected, it'll only be temporary.

Click the _Macro_ menu > _Macros..._ select _RunPipOnWin.py_ from the list and click _Execute_
In the dialog box requesting the module name type:
`numpy`
and click OK, it'll take some time and a message will appear in the Report View after a while such as _Successfully installed numpy_

Repeat running the macro and in the dialog box requesting the module name type:
`scipy`
and click OK and wait for the message in the Report View

Restart FreeCAD and hopefully the errors will have been cleared.

[RunPipOnWin.py.zip](https://github.com/user-attachments/files/16083132/RunPipOnWin.py.zip)

The only explanation as to how you got in this situation is another Addon you previously installed required a version of numpy that wasn't syncronised with the standard version of scipy included in FreeCAD. Think of the two as stablemates and should be upgraded together. @tomate44 if you've anything to add by all means let me know.

4567Nickam, 2024-07-03:
When I rename it still it doesn't show up under macros

Syres916, 2024-07-03:
Are you sure you've cleared the dot off the end? Notice how Windows shows it as a `PY File` in Explorer, here's mine, ignore the folder, I use non-standard location:

[image]

Syres916, 2024-07-03:
Also you do have file extensions being shown in Explorer as per https://support.microsoft.com/en-gb/windows/common-file-name-extensions-in-windows-da4a4430-8e76-89c5-59f7-1cdbbc75cb01 ?

4567Nickam, 2024-07-03:
Mine wasn't checked

tomate44, 2024-07-03:
> @tomate44 if you've anything to add by all means let me know.

Thanks for your help. I don't use and don't know anything about Windows.

4567Nickam, 2024-07-03:
Everything seems to be working fine. Thanks a lot.....much appreciated

closed 2024-07-04 by tomate44