tomate44/CurvesWB #125: Curves work bench not working on freecad

4567Nickam, 2024-07-03:
Kindly help. Curves workbench not working on freecad 0.21.2

JPT77, 2024-07-24:
For me it does not work as well. 
hope this helps
thank you

console output on open curves workspace:

    /usr/lib/freecad/Mod/Draft/importSVG.py:416: SyntaxWarning: invalid escape sequence '\.'
      _num = '([-+]?[0-9]*\.?[0-9]+([eE][-+]?[0-9]+)?)'
    /usr/lib/freecad/Mod/Draft/importSVG.py:732: SyntaxWarning: invalid escape sequence '\d'
      inks_ver_pars = re.search("\d+\.\d+", inks_full_ver)
    /usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
      _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
    /usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
      _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
    /usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
      _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
    /usr/lib/freecad/Mod/Draft/importSVG.py:938: SyntaxWarning: invalid escape sequence '\.'
      _num = '[-+]?[0-9]*\.?[0-9]+'
    /usr/lib/freecad/Mod/Draft/importSVG.py:1620: SyntaxWarning: invalid escape sequence '\('
      _val = '\((.*?)\)'
    /usr/lib/freecad/Mod/Draft/importSVG.py:1621: SyntaxWarning: invalid escape sequence '\s'
      _transf = _op + '\s*?' + _val
  
console output on extending bezier curve: 
  
      Part::FeaturePython: Link(s) to object(s) 'Sketch' go out of the allowed scope 'ExtendedCurve'. Instead, the linked object(s) reside within 'Body'.
    Traceback (most recent call last):
    File "/home/jan/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/curveExtendFP.py", line 50, in execute
        curve = curveExtend.getTrimmedCurve(edge)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/jan/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/curveExtend.py", line 13, in getTrimmedCurve
        c = edges[0].Curve.copy()
            ^^^^^^^^^^^^^^
    <class 'Base.FreeCADError'>: Unknown exception while reading attribute 'Curve' of object 'TopoShape'
    Recompute failed!

tomate44, 2024-08-13:
The first set of errors is not related to Curves workbench.
The second error should be fixed in version 0.6.44