tomate44/CurvesWB #130: Fill faces on Sketch On Surface causes list index out of range exception and fails.

ofmumu, 2024-08-21:
0.6.44

Following this tutorial: https://www.youtube.com/watch?v=th4eQmskTnc

At the point where I enable Fill Faces, this error is generated:

```
18:04:22  mapping shape #  1
18:04:22  Total : 72 shapes
18:04:22  Traceback (most recent call last):
  File "/Users/davidbrady/Library/Application Support/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 358, in execute
    ruled = Part.makeRuledSurface(shapes_1[i].Edges[j], shapes_2[i].Edges[j])
<class 'IndexError'>: list index out of range
18:04:22  Recompute failed!
```
[Spotlight.FCStd.zip](https://github.com/user-attachments/files/16684168/Spotlight.FCStd.zip)

PalaIkeLabs, 2024-10-07:
Fixed for me with 0.6.48

tomate44, 2024-10-12:
Fixed

closed 2024-10-12 by tomate44