tomate44/CurvesWB #131: Mapping circle to cone and setting thickness doesn't work right

theosib, 2024-08-22:
Steps:
- Install the Curves workbench
- Go to Part workbench
- Create a cone; leave its values at default
- Select the side of the cone
- Go to Curves
- Select Sketch on surface
- Open the sketch
- Draw a circle in the middle of it
- Go back to the Sketch_On_Surface object
- Set the thickness to 1

Expected:
I should see something cylinder-like sticking out of the cone.

Actual results
It's all twisted up in ways that only an image could convey.

Please also see the FCStd file attached. Also see: https://github.com/FreeCAD/FreeCAD/issues/16013

[image]

[curve sketch on cone.FCStd.zip](https://github.com/user-attachments/files/16719509/curve.sketch.on.cone.FCStd.zip)

theosib, 2024-08-22:
Setting the offset to something other than zero makes it work right.

theosib, 2024-08-22:
Someone on Discord can get this to work with an offset of zero. But I can't. He and I are both using dev version 38459. I'm betting he and I are on different platforms (I'm on Mac), and compiler optimizer differences are resulting different floating point rounding errors.

theosib, 2024-08-22:
Changing the offset doesn't fix everything. Here's a sketch that doesn't "thickness" right no matter what. 
[sketch on cone.FCStd.zip](https://github.com/user-attachments/files/16719819/sketch.on.cone.FCStd.zip)

theosib, 2024-08-22:
And here's another that doesn't come out right.
[sketch on cone2.FCStd.zip](https://github.com/user-attachments/files/16719845/sketch.on.cone2.FCStd.zip)

tomate44, 2024-09-04:
This should be fixed in 0.6.47
Please re-open and comment if it is not.

closed 2024-09-04 by tomate44