tomate44/CurvesWB #133: Can't map curve to Draft Clone of sketch or Link to sketch

theosib, 2024-08-30:
I have a sketch that I want to map to multiple different surfaces with different curvatures. It seems like either Draft Clone or a Link should do the trick. I have tried both and set the attachment to my curved surface, and when I click "Sketch on surface," it says "Please select a face (in the 3D view) or a sketch."

tomate44, 2024-09-04:
FreeCAD Version ... Example file

tomate44, 2024-09-04:
So you want to map a unique sketch to various target faces ?
And to achieve this, you want to create links of a unique Mapped_Sketch ?
This doesn't make much sense because the target faces probably have different parametric spaces.
So each target face need its own Mapped_Sketch.
The best solution is : 
- Select the target face and create a Sketch_On_Surface object
- Repeat for other target faces
- Create a unique sketch on the default XY plane with the shape you want to map
- Expand the Sketch_On_Surface objects, and select all the Mapped_Sketch objects in the Tree View
- Edit the **ExtraObject** property and point it to the Sketch containing the shape to map.
- This way, you still have the ability to adjust the blue rectangle of each Mapped_Sketch individually

closed 2024-10-12 by tomate44