tomate44/CurvesWB #134: joinCurves: Change sketch, join doesn't update

theosib, 2024-09-02:
I created a sketch where I made a wire from a line and an arc. I then used joinCurves to turn them into a single wire. The joinCurve itself lists the edges it's made from, suggesting that it has maintained a reference to them. So I then modified the original sketch, assuming that doing so would propagate to the joinCurves object. However, it did not. There also doesn't seem to be any option to request a recompute, and the builtin "recompute object" feature doesn't work on this.

If I knew in advance exactly what the wires need to look like, I'd have already made them the correct lengths. However, to get the visual effect I want, I need to tinker a bit. There is a complex chain of dependencies on these wires. If I have to recreate the whole tree every time I want to make a little tweak, this is going to be unmanagable.

tomate44, 2024-09-04:
FreeCAD Version ... Example file ... Error messages

theosib, 2024-09-05:
Looks like I made a mistake. I'm making ruled surfaces from various curves, and it seems that the problem is the ruled surface not updating, even from just wires directly in a sketch.

Sorry about that.

theosib, 2024-09-05:
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44

[knot plate 4 test.FCStd.zip](https://github.com/user-attachments/files/16898398/knot.plate.4.test.FCStd.zip)

Please see the attached design. What we're looking at is Ruled_Surface007, which is made up of JoinCurve002 and JoinCurve003. Those curves come from Sketch005.

To see the problem, go into Sketch005 and change the 45 degree angle to something else, like 60. When you exit the sketcher, it should significantly distort one of the bent knot curves adjacent to the corner, specifically the one labeled "lead into corner", which has a Curves sketch mapped to a face of Ruled_Surface007.

Another to look at is "lead out of corner", which maps to Ruled_Surface005, which is made from JoinCurves that map to Sketch004. Change one of the angles in that, and you'll see another example of the JoinCurves not updating properly.

I also have multiple ruled surfaces that just use simple curves, and when I update the sketches, the ruled surfaces update. It's only the ones that use JoinCurves that silently fail.

Thanks!