tomate44/CurvesWB #137: Flatten Surface is skewed in Freecad 1.1.0 dev

wanderingwaldo2, 2024-09-17:
When using the flatten surface feature on a truncated conical surface the output was skewed.

wanderingwaldo2, 2024-09-18:
[image]
Here is a picture of the issue.

tomate44, 2024-09-20:
Do you have an example file please ?

wanderingwaldo2, 2024-09-20:
I have attached an example file with the skewed output.

tomate44, 2024-09-20:
Where ? I cannot find it.

wanderingwaldo2, 2024-09-20:
[FlattenFailExample.zip](https://github.com/user-attachments/files/17078983/FlattenFailExample.zip)

Looks like it wasn't a supported file type for upload.  This should work.

tomate44, 2024-09-20:
Did you update CurvesWB to latest version before posting this bug report ?
For me, Flatten Face works fine with your file.
```
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38782 (Git)
Build type: Release
Branch: main
Hash: 8b9f5bdc4f4bd3cf4a440365fcc7cbcd2afe8580
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.8.2.dev
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * CurvesWB 0.6.47

```

wanderingwaldo2, 2024-09-20:
Sorry, I had the addon manager set to automatically update and thought that happened in the background, but once I opened the addon manager just now it updated and the flatten face command works.

tomate44, 2024-09-21:
Fine. Good news.

closed 2024-09-21 by tomate44