tomate44/CurvesWB #142: Update mismatch

MarcEngrie, 2025-01-23:
Hi,

As per your advise, I recently upgraded to Curves WB 0.6.4 (to solve array on Sketch_on_surface issue). 

So when I look in Addon Manager, I see this 

[image]

Clicking on it show this

[image]

Also, FreeCAD shows an update is available 

[image]

Clicking on it, runs the update proces but, od course, nothi,h happens as I think downgrading is not an option or because 0.6.4 is higher then 0.6.3. 

Anyway, strange and confusing. 

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Belgium (en_BE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.54
  * dodo 1.0.1
  * fasteners 0.5.36
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.3.0
  * LCInterlocking
  * sheetmetal 0.7.4
  * woodworking 0.21.2.33771.0.23.1

tomate44, 2025-01-25:
Try clearing the cache of the addon manager.
Or try de-install / re-install the WB.
Anyway, I don't think this is a CurvesWB bug.

closed 2025-01-25 by tomate44

MarcEngrie, 2025-01-25:
de-install / re-install the WB was the thing to do. 
Thx