tomate44/CurvesWB #143: When referencing Extra Objects, throws list index out of range exception

theosib, 2025-01-26:
I'm trying to map a design to a ruled surface, and I'm getting errors, and it's not mapping correctly.

The patten is like this:
[image]

But when mapped to a surface, it looks like this:
[image]

And it's throwing these errors:
```
20:54:52  pyException: Traceback (most recent call last):
  File "/Users/theosib/Library/Application Support/FreeCAD/Mod/Curves/./freecad/Curves/Sketch_On_Surface.py", line 359, in execute
    if not (len(shapes_1[i].Edges) == len(shapes_2[i].Edges)):
                                          ~~~~~~~~^^^
<class 'IndexError'>: list index out of range
20:54:52  Sketch_On_Surface: list index out of range
```

I'm also attaching my design file.
[fmrd mold.FCStd.zip](https://github.com/user-attachments/files/18548604/fmrd.mold.FCStd.zip)

I'm using this version of FreeCAD:
```
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

In general, Curves WB seems to have a lot of trouble mapping to curved surfaces where if often fills in spots that should be empty and also can't apply a thickness.

theosib, 2025-01-26:
Oh, and if I change the Extra Objects in any way, recomputing doesn't update the object.

tomate44, 2025-01-28:
Your input Sketch is self-intersecting.
This is forbidden, as in any other padding / extruding operation in FC.

closed 2025-01-28 by tomate44

theosib, 2025-02-09:
This particular one works fine otherwise. It just doesn't work in Curves.