tomate44/CurvesWB #144: Map sketch on surface does not work correctly with cones

wolflu05, 2025-01-27:
When using the "Sketch on Surface" tool on the surface of a cone, it gets unwrapped as a rectangular shape. So when drawing a square/circle/... it gets distorted on the cone. I would expect that a cone unwraps correctly (like when you use the "Flatten Face" option). Same for a blunt cone.

[image]

[image]

This is what I would expect (for a blunt cone, the top point would be also an arc like on the bottom):

[image]

I'm currently using FreeCad in version:
```
OS: macOS 15.1.1
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.54
```

tomate44, 2025-01-28:
Sketch_On_Surface work as expected.
But I agree that using the flatten surfaces as transfer surface in Sketch_On_Surface (instead of the blue rectangle) would be a great enhancement.
I tag this as a Feature Request.
Thank you.

wolflu05, 2025-01-29:
Thank you for your quick response, that would be awesome.

Do you have any idea for now how I can solve such a problem without that feature?