tomate44/CurvesWB #156: Faces of complex sketch filled incorrectly

kalbfuss, 2025-05-01:
I am trying to map a drawing (sketch) to a curved surface. The mapping works in principle, but faces enclosed by the paths are not filled correctly. This is the sketch I am trying to map:

[image]

As far as I can tell, all the paths are closed. If I use the "Make face from wires" tool in the Part workbench, the result is as expected:

[image]

When I map the sketch to my curved surface, the faces between the paths are no longer filled correctly, but spill over:

[image]
[image]

This is the FreeCAD file I am using and which can be used to reproduce the problem:

[Engraving.zip](https://github.com/user-attachments/files/20003591/Engraving.zip)

And here is my version information:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * lattice2 1.0.0
  * Curves 0.6.51
```

Any help appreciated. I am happy to do more testing as instructed.

kalbfuss, 2025-05-02:
I have done a bit of research in the meantime. I assume the problem is that my sketch contains closed paths within closed paths. It is thus not sufficient to fill the area between the paths. In addition, the area spanned by the inner closed paths must be subtracted from the area spanned by the outer closed paths. Otherwise the result looks pretty much like what the Curves workbench produces currently.

luzpaz, 2025-07-11:
@tomate44 is there a way you want to track this ticket with regards to the WIP code refractor ? Perhaps give it a label ?