tomate44/CurvesWB #166: Gordon: unsupported operand type(s) for *: 'NoneType' and 'int'  and FreeCAD freezes

razorree, 2025-07-11:
When I try to create Gordon Surface

`Traceback (most recent call last):
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordonFP.py", line 144, in execute
    s = gordon_surf.surface()
        ^^^^^^^^^^^^^^^^^^^^^
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 361, in surface
    self.perform()
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 322, in perform
    self.make_curves_compatible()
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 558, in make_curves_compatible
    self.guides[spline_v_idx] = bsa.reparametrizeBSplineContinuouslyApprox(guide, oldParameterGuide, newParametersGuides, max_cp_v)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineAlgorithms.py", line 644, in reparametrizeBSplineContinuouslyApprox
    result, error = approximationObj.FitCurveOptimal(parameters, 10)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineApproxInterp.py", line 165, in FitCurveOptimal
    old_error = error * 2
                ~~~~~~^~~
<class 'TypeError'>: unsupported operand type(s) for *: 'NoneType' and 'int'
Gordon: unsupported operand type(s) for *: 'NoneType' and 'int'
`

razorree, 2025-07-11:
`<Sketch> SketchObject.cpp(9132): External geometry logi_g_pro#Sketch006.e1 missing reference: Sketch005.;g34;SKT
Traceback (most recent call last):
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordonFP.py", line 144, in execute
    s = gordon_surf.surface()
        ^^^^^^^^^^^^^^^^^^^^^
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 361, in surface
    self.perform()
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 322, in perform
    self.make_curves_compatible()
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 558, in make_curves_compatible
    self.guides[spline_v_idx] = bsa.reparametrizeBSplineContinuouslyApprox(guide, oldParameterGuide, newParametersGuides, max_cp_v)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineAlgorithms.py", line 644, in reparametrizeBSplineContinuouslyApprox
    result, error = approximationObj.FitCurveOptimal(parameters, 10)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rzr/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineApproxInterp.py", line 169, in FitCurveOptimal
    while ((error > 0) and ((old_error - error) / max(error, 1e-6) > 1e-6) and (iteration < maxIter)):
            ^^^^^^^^^
<class 'TypeError'>: '>' not supported between instances of 'NoneType' and 'int'
Gordon: '>' not supported between instances of 'NoneType' and 'int'
`

closed 2025-07-12 by tomate44

razorree, 2025-07-12:
I suspect that the project isn't perfect, i'm just learning, however messages/errors/warnings are not very clear.

and of course those design errors shouldn't cause the program to crash or freeze.

razorree, 2025-07-12:
FC SHOULDN'T freeze and require KILL  -  that's minimum expected behaviour.

(this issue is not about creating wrong Gordon Surface)

tomate44, 2025-07-12:
> FC SHOULDN'T freeze and require KILL - that's minimum expected behaviour.

That's minimum expected behaviour for a **stable** software.
As clearly stated in the README, Curves Wb is experimental, and will probably always be.
It is just my personal playground. People are free to give it a try, at their own risk. No other promises.

razorree, 2025-07-12:
ah, sorry, good point :) didn't realise, as Curves Workbench is everywhere in FC documentation and YT.

tomate44, 2025-07-12:
It became quite popular because it meets a need for surfacing workflow.
Unfortunately, I have limited time for this project, that I want to spend on learning python coding, geometry and experimenting. Almost no documentation, few user support.
I am aware this is not a very professional workflow. But that's all I can offer.
FYI, it is usually better to post on the CurvesWB [dedicated thread](https://forum.freecad.org/viewtopic.php?t=22675) on the FC forum. There are a few people there that know the workbench well and are able to help.